segalion/raspipool #1: Some questions

sunflowerABB, 2019-09-10:
Very interesting project, thanks for sharing your work!
A few questions:
1. could I use your code with a home assistant instance running remotely on another server? What would have to be changed for the sensor values to be forwarded to the remote server? I noticed you defined the sensors in the ORP.yaml file:

`- platform: atlas_scientific
   port: /dev/ttyUSB0
   offset: 78
   scan_interval: 31536000`

2. Since the Raspi wouldn't be doing much besides forwarding sensor values, do you have any suggestion on how to adapt your code to run on an Arduino instead, e.g. using ESPhome?

3. Adding the Salt Water Chlorinator feature soon would be awesome, please! 👍

segalion, 2019-09-11:
1. Runnig HA remotely could be possible, but makes things more complicated, and less reliable. You allways need a RPI to connect sensors to GPIOs, and UARTs, (remote GPIO and/or something to extend UART over TCP...)
2. I was thinking in use tentacle board too. Maybe in the future I could buy one and develope a version for I2C protocol for the custom component.
3. Yes... thats one of the first thing I were been thinking a lot to decide where to build the project: arduino/ESP vs RPI, and finally I decided for a RPI for three main reasons:
- Mobile notifications (mutch better with RPI+HA).
- Better outside-integrations (i.e. weather, google calendar, etc.)
- Further integrations (presence detections, cameras and microphones, speech-to-text for commands, music, ...)
So then, its not easy to replicate HA yaml config on a ESP/arduino code...
4. I don´t have SWC so is difictult for me know the requirements...
At first aprox. you can connect the bleach injection relay to start/stop the SWC, and configure raspipool with "bleach concentration = 1.75%" and then aproximate the "gr of Clorine/hour" of your SWC for the same ml/min.

MimbaMonkeyHouse, 2019-09-27:
@sunflowerABB, you could always have this running in one Home Assistant instance and use the mqqtt publish service to forward sensor data and even switches to another instance on a different computer...

tobiasneumann42, 2020-06-21:
had the same issue, my home assistant is running on VMware and the Pi with the connected sensors located in a housing close to the pool. As suggested by @MimbaMonkeyHouse installed mosquitto mqtt broker on Pi. Python script on Pi sends sensor readings (have several temperature sensors, pH, ORP and pressure sensors) locally to the broker. Home assistant subscribes to these readings. Works pretty straight forward and stable.

lucky79, 2021-01-15:
@tobiasneumann42 Would you be willing to share more details please? I am trying to do the same, so it would help me a lot. Right now I only need it for pH and ORP, temps from 1wire sensors connected to UniPi Lite board and also relays are possible using the rest api. Thank you!

V4l3n10, 2021-02-16:
> 1. Runnig HA remotely could be possible, but makes things more complicated, and less reliable. You allways need a RPI to connect sensors to GPIOs, and UARTs, (remote GPIO and/or something to extend UART over TCP...)
> 2. I was thinking in use tentacle board too. Maybe in the future I could buy one and develope a version for I2C protocol for the custom component.
> 3. Yes... thats one of the first thing I were been thinking a lot to decide where to build the project: arduino/ESP vs RPI, and finally I decided for a RPI for three main reasons:
> 
> * Mobile notifications (mutch better with RPI+HA).
> * Better outside-integrations (i.e. weather, google calendar, etc.)
> * Further integrations (presence detections, cameras and microphones, speech-to-text for commands, music, ...)
>   So then, its not easy to replicate HA yaml config on a ESP/arduino code...
> 
> 1. I don´t have SWC so is difictult for me know the requirements...
>    At first aprox. you can connect the bleach injection relay to start/stop the SWC, and configure raspipool with "bleach concentration = 1.75%" and then aproximate the "gr of chlorine per hour" of your SWC with same "ml/min".

@segalion, Have you ever had the chance to develop a version for I2C protocol for the Tentacle?