segalion/raspipool #3: Help for Install

andyyyy2306, 2019-10-02:
Hi, i try to install your Software on an Rpi and this is what i get 
[image]
could you help me pls ?

segalion, 2019-10-03:
Sorry for the delay.
Seems that there are a problem with your lovelace config. Lovelace defines the frontend (how the information is distributed in the web access)
Has you copied the 'ui-lovelace.yaml' in the base config directory (same where 'configuration.yaml' is placed)?
Has the option 
```
lovelace:
  mode: yaml
```
inside the 'configuration.yaml' file?

andyyyy2306, 2019-10-04:
Hi, thanks for your answer. That works !!
I am new in Home Assistent. work for yaers with Domoticz.
So i try translate to german. It´s right to set the names in each .yaml file ?

segalion, 2019-10-07:
Thanks @andyyyy2306 
I am pleased to hear someone trying to implement raspipool. 

There is no simple way to handle multiple languages ​​in HA outside the inner components. But I recommend to not change the yaml files, in order to make new updates can work easily. Please, use change the "friendly name" from the web, for each entity, and all traslation will be stored in a json file in /home/homeassistant/.homeassistant/.storage/core.entity_registry 

Could be great to has this file traslated to german!!!, this could be an aproximation to a basic "internationalization" aproach...

About calibrate sensors, there are two ways:
1. Hard way: conenct to the ezo circuit by console, and callibrate it followinf Atlas Sccientific instructions.
2. Simple way: Configure an 'offset' parameter to the orp or ph yaml config file

andyyyy2306, 2019-10-07:
HI @segalion ,
thanks for your help !!

> Please, use change the "friendly name" from the web, for each entity..
thats my Problem, the register is emty ..
should there listed all entity´s you use in your automations ?

[image]

segalion, 2019-10-09:
@andyyyy2306 
Sorry about that... Seems that with 'ui-lovelace.yaml' cannot change friendly-name from gui.
Home assistant is making a lot of changes between yaml mode and json (more automatic) mode... 

Please, in order to traslate, could you try to change the 'ui-lovelace.yaml' "name" parameters?
At least you change only one file.

sralesurk, 2019-12-30:
Hi @segalion 
You project looks great. i'm trying it out, since this would be great for my new pool.
I'm completely new to HA, have used Openhab before. I'm considering changing to HA and rapipool ;-)
I'm having a bit of problem getting things up and running (no sensor hardware yet)
The HA stops working on the RPI after i made the changes in your description.
I believe my problem is secrets.yaml (step 3), I'm not sure if im doing this correctlyI , could you perhaps provide screen shoot of your secrets.yaml (with your secrets crossed out)
Thanks for a great project, 
i'm also looking to changing the setup a bit to include a SWC