segalion/raspipool #4: Fluids dynamics

metropt, 2019-11-03:
How did you design the bypass Tube dimensions? I suppose you need to guarantee some pressure between the main pumping system and the bypass, is that right?

Thank you

segalion, 2019-11-08:
I dont understand your question. The bypass is carefully designed to:

1. get a consistent flow (high enough) It is garanted because it is in recirculating mode, so a little part the main returning flow (from filter) is going to be reabsorbed by the pump, to by-pass, and go again to the filter.
2. get an adecuate flow (low enough). It is garanted by the 20D pipes instead 50 or 65D main
3. the water recirculating is always filtered before measurement, to improve the lognevity of the probes
4. easy maintenance: outiside the main flow, with 20D valves, you can cut the bypass for maintenance (change probes, callibrate, etc.)
5. allways wet. To mantain the probes inside water, the bypass need to be at the most bottom, over a 50D pipe (more water, more difficult to evaporate).

fpalinf, 2019-12-10:
It would be nice to include those rationale on the wiki.
I was about to question the design after reading the https://github.com/segalion/raspipool/wiki/Bypass-for-sensors
 but it is pretty clear with your explanation.
Thanks again for your work ! Amazing !

socbrian, 2020-05-11:
I am having a hard time getting my head around how new water gets to the probes. The image [here](https://raw.githubusercontent.com/segalion/raspipool/master/help/media/raspipool_overview.png) seems like the sensors will get "dirty" water before the pump and then never have pressure because it is before the pump.

0u751d3r, 2020-07-29:
>  seems like the sensors will get "dirty" water before the pump and then never have pressure because it is before the pump.
If you look at the diagram, the "sensor loop" is connected to both the suction AND pressure side of the pump (the latter is after the filter). Water will flow from the higher pressure toward the lower pressure, i.e. it will flow from the pump outlet back to its inlet, so it will always be clean water.

segalion, 2020-10-07:
> If you look at the diagram, the "sensor loop" is connected to both the suction AND pressure side of the pump (the latter is after the filter). Water will flow from the higher pressure toward the [...]

You are right. Probes are getting "clean" water (after the filter), as it is a recirculation system.
Note that this bypass has to be as low as possible to ensure it never runs dry (and probes "caput").

segalion, 2020-10-07:
I close this issue as it is solved by @0u751d3r

closed 2020-10-07 by segalion