segalion/raspipool #5: Hi, I have a few questions

justinladwig, 2019-12-01:


MimbaMonkeyHouse, 2020-01-06:
I also would love to know the answers for questions 1 & 2. @segalion can you please chime in.

For question 3: Nope, I would not recommend using a Pi Zero to host any Home Assistant instance. It is actually not recommended on the documentation for Home Assistant...

justinladwig, 2020-01-14:
Hello again, 

could you tell me how you connect the second atlas scientific sensor to the Raspberry Pi? On the sensor connection shema you connect it to Pins 36 and 37, i.e. GPIO 16 and 26. 
As far as I know, the Raspberry Pi should only have a UART on pins 8 and 10, i.e. GPIO 14 and GPIO 15, which the other sensor is already connected.

I hope you'll reply.
Justin

Jackk16, 2020-06-03:
Hi Segalion,

Very promising solution!
I am getting prepared to try your solution for a automated pool control.  Like the other questions its hard to find how to connect the peristaltic pumps. Could you point out how that works, perhaps add it to the connection diagrams?

Regards,
Jack

segalion, 2020-10-07:
Sorry, I think too I have to work more on documentate the injection part.

Even more, I think I have to make a global rewrite of all project. Could be great to include some hard-internal-parts passed to a custom component core, to change the illegible jinga2 parts for elegant python code.

segalion, 2020-10-07:
> Hello again,
>
> could you tell me how you connect the second atlas scientific sensor to the Raspberry Pi? On the sensor connection shema you connect it to Pins 36 and 37, i.e. GPIO 16 and 26.
> As far [...]

RPI2-3 has a second "limited" UART. To enable it, you have to...

Edit rpi config file with:

sudo nano /boot/config.txt

including this:

force_turbo=1
enable_uart=1
dtoverlay=pi3-disable-bt
# sencond UART1 /dev/ttyS0 
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33

Take into account that with rpi3, bluetooth uses the UART, so you have to disable it.

I think tha RPI4 has even 4 complete UARTs.