segalion/raspipool #6: ec probe not working

docsparks, 2020-04-02:
using an ezo usb board for ph, and another ezo usb board for ec, my rpi3 running hassio  shows both sensors in hardware but only “unknown” for ezo_conductivity.   not sure how to troubleshoot this but the usb ezo and ec probe work fine with windows.

segalion, 2020-04-28:
I havent EC board, so I cant figure how this card is identified. Sensor module try to discover based on this:
```
        ezos = {"ph": ['ph', 'pH', 'mdi:alpha-h-circle'],
               "orp": ['orp', 'mV', 'mdi:alpha-r-circle'],
               "or": ['orp', 'mV', 'mdi:alpha-r-circle'],
               "do": ['dissolved_oxygen','mV', 'mdi:alpha-x-circle'],
               "d.o.": ['dissolved_oxygen','mV', 'mdi:alpha-x-circle'],
               "ec": ['conductivity', "EC", 'mdi:alpha-c-circle']}
```
that I found on the EZO instructions. 

But I have seen that there are changes in the identification (i.e. D.O. or do). Maybe with 'ec' happens the same...
Please could you try changing sensor.py code with:

`               "e.c.": ['conductivity', "EC", 'mdi:alpha-c-circle']}`

docsparks, 2020-04-29:
I meant to update my comment.     Unless the EC probe is configured to only give back a numeric value, around line 110 of the sensor.py file, there is no error handling for any comma delimited data.   The EC probe gives multiple options for output, and the user must have the EC probe setup right to make the code not error out.    Thank you again for your work.

`       r = self._read()
            #self._state = float(r) + self._offset
            self._state = (r)`