segalion/raspipool #9: EZO Carrier board : isolated version or not ?

Nickous, 2020-04-28:
Your work is awesome. Thank you so much for sharing it here.
I wonder about the EZO carrier bord...
Your link is about the non-isolated version. Even if you have more than one sensor (PH and ORP)?
Does the electrically isolated board worth the cost (40$ vs 19$!!) ?
Thank you for your feedback

segalion, 2020-04-28:
I am working with non-isolated boards. But I have to say I had a lot of problems with noise from ethernet card and bad PSU and usb power cable.

With a good quality PSU and cable, and no LAN connection I have good meassurements for pH and ORP.
Without isolation, you cant use ethernet. Only wifi.

The official information says that other measurements (I cant remember if DO or EC, isolation is obligatory).

Nickous, 2020-04-28:
As I intend to use lan connexion, isolation is more than recomanded
Thanks @segalion for the fast reply and advice

closed 2020-04-28 by Nickous