segalion/raspipool #10: FC from ORP and PH calculation

huehueteotl, 2020-05-14:
First of all thanks for a great project which I used for my own fork!
I use it with a Tentacle HAT from whiteboxes.ch, and modified your code to enable I2C (I can send you a pull req is you are interested; it should work for both serial and I2C sensors).

My question: what formula do you use for the estimated FC calculation from the ORP and pH values?
Of course I see it in your yaml but I'm having a hard time matching that to the logarithmic chart you have in the wiki, and what I can find online (I know all about FC != f(ORP+PH) now... ;-)).
Can you tell me the source of the info you used?

Nickous, 2020-05-22:
Hi @huehueteotl 
I'm about to choose pH and ORP module card.
I'm very interested in your solution with I2C protocol (and Tentacle T3 from whiteboxes.ch).
Is-it possible to share it?

huehueteotl, 2020-05-22:
Hi @Nickous 
I have just committed my changes in my fork (https://github.com/huehueteotl/raspipool/commits/master) so you can see the changes.

It already works but there is (as usual) still some work to do: I also have two Atlas Ezo-pumps for pH and Cl liquids which I still have to connect which also use i2c.
Have fun ;-)

Nickous, 2020-05-29:
> Hi @Nickous
> I have just committed my changes in my fork (https://github.com/huehueteotl/raspipool/commits/master) so you can see the changes.
>
> It already works but there is (as usual) still some [...]

Thank you very much for your commits @huehueteotl .
It's really helpfull. I'm new in HA so I have so much to understand...

rzulian, 2020-06-05:
> First of all thanks for a great project which I used for my own fork!
> I use it with a Tentacle HAT from whiteboxes.ch, and modified your code to enable I2C (I can send you a pull req is you are [...]

I was looking for the same info. I think it's important also for safety matters that we document the source. I've understood also that he is working on thecCorrection of FC-ORP based on CYA.

ptomasroos, 2020-06-15:
Great work @huehueteotl making the HAT work.

I can definitely help out on going with your branch as I've bought the pieces my self and are about to get a system up and running with the same components as you!

virgilm, 2020-07-20:
I am looking at the same solution (I2C based), I can help too. I forked my own [repo](https://github.com/virgilm/raspipool) from https://github.com/huehueteotl/raspipool/commits/master, as this one seems not maintained anymore.

Nice project!

nibi79, 2020-07-25:
What is the current status of your forks?
Does anyone using  Ezo-pumps for pH and Cl now?

ptomasroos, 2020-07-25:
I haven’t started yet but starting to assemble the sensors

virgilm, 2020-07-26:
@huehueteotl - have you had to do pH sensor calibration? I am trying to do it via I2C, but can't figure out how. In particular, I would like to do it in the context of Home assistant. Any ideas?

@nibi79 I pushed some fixes to my repo, I'd be happy to push it back to this project or use mine/@huehueteotl's as a place to collaborate.

I have the whole PI + T3 + Sensors up and running.

huehueteotl, 2020-07-26:
@virgilm I calibrated my pH and ORP sensor on one of the old RPi2's I have lying around, using https://github.com/AtlasScientific/Raspberry-Pi-sample-code/blob/master/i2c.py

An option in HA would be nice, but I also have a lot to learn about YAML/Lovelace to be able to fix that ;-)

huehueteotl, 2020-07-26:
@nibi79 Ezo-pumps are configured for i2c but still laying around to be connected. It also requires quite some changes to the custom sensor.py to send the right commands to the pump. No time yet to play with that :-(

This being the first time we own an actual pool (it came with the house we recently bought) there is a lot to find out! I just finished replacing the big pool filter pump for the second time (it has a Filtrinov monoblock installed) which was a real challenge.

virgilm, 2020-07-29:
@huehueteotl - I merged a [PR](https://github.com/virgilm/raspipool/pull/9) to do sensor calibration. A bit awkward, due to me being new to HA and HA being... incomplete, but it works. Please check it out if you have time and feel free to improve it. I'd be curious to see if works for you too. I will continue to work on my repo, feel free to open issues there if needed.

nibi79, 2020-08-16:
Is it possible to mount the relay and the tentacle both on the Pi?

huehueteotl, 2020-08-17:
> Is it possible to mount the relay and the tentacle both on the Pi?

sure, it is stackable and exposes all the pins on top of the board.
see https://www.whiteboxes.ch/wp-content/uploads/2016/12/143.png

[image]

urbatecte, 2020-10-06:
Hi there !
Ouch, the formula is in the fc.yaml file [...](https://github.com/segalion/raspipool/blob/master/packages/raspipool/fc.yaml)
       value_template: "{{ ( 0.23 * (1 + states('input_number.e_fc_adjust')|float / 100 ) * ( 14 - states('sensor.ph_mean')|float) ** ( 1 / (400 - states('sensor.orp_mean')|float) ) * ( states('sensor.ph_mean')|float -4.1) ** ( ( states('sensor.orp_mean')|float - 516)/145) + 10.0 ** ( (states('sensor.orp_mean')|float + states('sensor.ph_mean')|float * 70 -1282) / 40 ) ) |round(1) }}"

I need to dig into that ...

Thanks a lot @segalion for your great work !

huehueteotl, 2020-10-07:
Hey @urbatecte 
Surely I saw the formula in the yaml ;-)
My question is where he got it from as there does not really seem to be an easy/straight relation between PH+ORP and FC.
It is just an approximation, which is fine, but in my case I do not get any values that make sense:
`(0.00027968614653875164+0.0015014854238866076j) ppm`
Must be something with the constants he is using but what?

segalion, 2020-10-07:
First of all, sorry about the low support on last year.

You will not be able to find it anywhere, since it is its own creation (you can call it "Segalion´s formula"), trying to "emulate" 
https://github.com/segalion/raspipool/wiki/FC-vs-ORP-and-pH
of recognized tables like: 

https://www.servovendi.com/media/catalog/product/cache/b220f734dc818b42f97889a5ce33e7cb/c/o/correlation_between_ph_orp-redox_mv_and_free_chlorine_ppm__1_5.png
https://chemtronicsindia.com/images/Chlorine-VS-ORP-graph.jpg

https://chemtronicsindia.com/images/Chlorine-VS-ORP-graph.jpg

https://www.yalitech.cl/media/wysiwyg/tabla_pH_orp.png

...

I believe that there is no function to obtain the free chlorine (FC) as a function of the ORP and pH, and this formula is a great approximation widely tested (for the typical wide-ranges in a swimming pool), both with the different empirical tables accessible publicly and from my own experience during two seasons.

Anyway, I am pleased to see that there are more people who value this formula as much as I do, as is the base for any FC regulated system based on ORP.

Take into account that thy are meassures for "typical" water. ORP can be affected for many other undetermined things, so maybe a proper calibration of FC-ORP, in every swimming-pool is a good recomendation.

huehueteotl, 2020-10-07:
Thanks for your reply, that is helpful info and those tables are not really easy to find; at least I didn't find them last time ;-)

My pH and ORP are perfect (7.37 and 742 currently) but the FC gives a weird value so I will double check if all the sensor values used in the template are set correctly.

Closing this ticket to prevent too much off-topic stuff.

closed 2020-10-07 by huehueteotl

cperricos, 2022-12-30:
Hello and thanks for this great project. I would like to use the formula in another piece of related python code, but nor sure I have interpreted the YAML code correctly. Is it:

( 0.23 * (1 + a / 100 ) * ( 14 - b)* ( 1 / (400 - c ) * ( b -4.1) * ( ( c - 516)/145) + 10.0 * ( (c + b)* 70 -1282) / 40 ) ) 

Where a is e_fc_adjust (not sure what that is)
b is PH Mean
c is ORP mean in mV

many thanks in advance!

huehueteotl, 2023-02-08:
Hi @cperricos 
Sorry for the extreme delay in my reply.
** = pow() so I believe the python version should be like this:
```python
# Some example values:
e_fc_adjust = 0;	# FC Adjustment in percentage
ph_mean = 7.3;		# Average PH value
orp_mean = 724;		# Average ORP value

e_fc = round( 0.23 * (1 - e_fc_adjust) * (14 - ph_mean) ** (1/(400 - orp_mean))*(ph_mean - 4.1) ** ( (orp_mean - 516)/145) + 10.0 ** ( (orp_mean + ph_mean * 70 - 1282 ) / 40 ), 1 );
print( "Estimated FC = ", e_fc );
```

cperricos, 2023-02-08:
Ah ok thanks very much!

mzakharo, 2023-02-12:
> Hi @cperricos Sorry for the extreme delay in my reply. ** = pow() so I believe the python version should be like this:
>
> ```python
> # Some example values:
> e_fc_adjust = 0;	# FC Adjustment in [...]

For anyone interested in building their own models of  ORP/PH -> ORP conversion, I found this chart helpful:
https://jenfitch.com/free-chlorine-orpmv-vs-ph-new/
You can find it in .csv form here: [orp](https://raw.githubusercontent.com/mzakharo/micropython/tubby/ports/esp32/orp.csv)

I 've put together a basic ML flow that converts this data to a model here: [build_model.py](https://github.com/mzakharo/micropython/blob/tubby/ports/esp32/build_model.py). It could use some more love, but it gets the flow going.

You can find prebuilt models here:
[model_fc.h5](https://github.com/mzakharo/micropython/blob/tubby/ports/esp32/model_fc.h5): model for predicting FC from  ORP/PH

[model_orp.h5](https://github.com/mzakharo/micropython/blob/tubby/ports/esp32/model_orp.h5): model for predicting ORP from FC/PH. This one can be used to calibrate your ORP sensor by getting your water lab tested

Basic [script](https://github.com/mzakharo/micropython/blob/tubby/ports/esp32/predict.py) to load and run these models. usage:

```bash
python3 ./predict.py  model_fc.h5 666 7.6
-> output:  0.83490694
./predict.py model_orp.h5 7.6 0.85
-> output:  667.04565
```