segalion/raspipool #11: issues with pH and ORP probes

tobiasneumann42, 2020-06-20:
good day, 

having some strange problems with pH and ORP probe, when running each of them standalone readings are correct. the moment I connect them both the readings on both sensors go all over the place (pH claims readings like 1,2 something or 9,7 something, same for ORP level going down to 40 - 50 range from 400 something). Used the non-isolated versions and have them connected as specified in the wiring diagram to raspberry Pi 4 using UARTs. 

checked the sensors with the basic python code provided by Atlas Scientific, no joy... (sensors are calibrated).

any ideas highly welcome, not sure what I am missing ...

viktor-c, 2020-08-07:
I think the problem is not in the code, but rather the fact that you are using unshielded sensors. This means that the readings get mixed up through the influence of the sensors on each other. You have to shield gnd or use a digital circuit.

helix0712, 2020-10-01:
Hi,

looking at the DFRobot example at https://www.dfrobot.com/blog-733.html we would probably need some of these:
https://www.dfrobot.com/product-1621.html to isolate signals from each other.

urbatecte, 2020-10-06:
Hi, good to read the hint.
I ran into the same issue, but I just went through new calibration (indeed it's more a characterization) to re-tune the pH reading with both probe in the water at the same time.
For pH it's seems to be OK (at least in the glasses with the pH powder). I'am using poor chinese pH/ORP probe as well as poor ph/orp meter electronic board. 
I'am only using ESP Home for now, and I need to go through the code of this fantastic project to get the ORP/pH/FC math formula/abacus o:)

segalion, 2020-10-07:
Thanks. 
Sure is a problem of "unshielded".
Please take into account that every connection to your system can make/traslate noise to the final probe...
1. Get the better PSU you can.
2. Get the better cable (USB) you can.
3. Dont connect any more things (NO ethernet, other sensors, etc.)
When you has good quiality (stable) measurements, you can try other sensors (i.e. I have a little USB mic connected, to remotelly hear the motor)

I have reach working fine pH and ORP with unshielded, in a raspberry pi. But only with good PSU, and no ethernet connection. If I connect ethernet, begin to huge noise on measurements (even more than +-0.5 pH!!!)

urbatecte, 2020-10-07:
Yeah, butI think that embedded ADC are quite sensitive to PSU noise.
I've then give up those ADC and switch to a [cheap ](https://fr.aliexpress.com/item/32825130324.html) ADS1115 and see quite great improvement.
Knowing that ESPHome support this chip it have been straightforward.

segalion, 2020-10-07:
> 
> 
> Yeah, butI think that embedded ADC are quite sensitive to PSU noise.
> I've then give up those ADC and switch to a [cheap ](https://fr.aliexpress.com/item/32825130324.html) ADS1115 and see quite great improvement.
> Knowing that ESPHome support this chip it have been straightforward.

Do you plan to create your own pH and ORP sensors with ESP32 device and ADS1115?
This could be really great, but its not easy... 

Some aproach...
https://www.e-tinkers.com/2019/11/measure-ph-with-a-low-cost-arduino-ph-sensor-board/