segalion/raspipool #14: Integration into existing HA Installation

jgthornburgh, 2020-09-10:
Is this able to be added into an existing installation of HA?

segalion, 2020-10-07:
Yes, following instructions...
I think now, project is some outdated (developed for 0.70-080 HA versions), and I dont know how it works on modern HA.
Sure it need a heavy revision...

nagyrobi, 2021-02-03:
As far as I can see it can be easily implemented to any existing HA, without even needing any raspberry.

All the sensors and the relays can be connected to any ESP8266-based device running [Tasmota](https://tasmota.github.io/docs/EZO/), to just simply forward the data in. The configuration only needs to be updated to handle sensors through different protocols (like MQTT instead of raspbi's own ports).
I personally don't like having a raspberry in a humid pool service place, having the sensors there is more than enough.

V4l3n10, 2021-02-16:
> As far as I can see it can be easily implemented to any existing HA, without even needing any raspberry.
>
> All the sensors and the relays can be connected to any ESP8266-based device running [...]

Guys, is there a way to integrate it using Raspbi's GPIO's?
I am asking as I have all the setup already and the sensors are connected via a Tentacle T3 Hat on the RPI.