segalion/raspipool #15: adding calibration options via service

docsparks, 2020-09-16:
in an attempt to extend your original service, i have failed to figure out how to use the service to modify temperature, which is crucial to my setup.

I have posted on the homeassistant forums here:
[https://community.home-assistant.io/t/adding-input-parameters-to-a-custom-component-that-is-only-a-sensor/227281](url)

and have found a significant improvement of your original atlasscienetific custom component here:
[https://github.com/virgilm/raspipool/blob/master/custom_components/atlas_scientific/sensor.py](url)

I hope this can get some light shed on it. thanks again for your work.

segalion, 2020-10-07:
Hello. ezo cards can be calibrated with 'tty' interation acording to their documentation.

From my experience for this type of project the temperature correction (in the pH) is minimal, and only necessary in industrial processes with extremely expensive probes where a decimal is important. 
I think that the calibration option for our cheap probes can be done perfectly with a Home Assistant sensor template without the need for anything else. Maybe could be great an UI to calibrate this sensor from HA lovelace.

About virgilm I2C modification is a great advance... I will try to include it (when spare time)....

docsparks, 2020-10-08:
my use isnt for pools to be exact, and water temp at 68 , vs 86, does make a fair difference.   especially within calibration. 
also my ezo ec probe has a different process to calibrate and set modes.

the ezo usb adapter is for sure my best use case, as i have a rpi0w, usb hub, and ph and ec usb ezo kits all assembled nicely with zero wiring and zero interface, just a few plug and play cables at minimal cost vs effort.  to use another  interface method is not an option in comparison.

i have looked into usb-serial integrations and other methods to interact with the sensors but i felt having such close access to the code, a firm understanding of programming, the only thing i lack here is the python programming context, and understanding of basics to use python.    i only asked for aid because you clearly are a few years ahead of me, and hoping to save some time.  

like to even finish the temperature compensation service you started to implement in the original files, would help me greatly be able to improve upon the coding.

thanks!

virgilm, 2020-10-08:
I wrote a (quite basic for now) sensor calibration (backend + UI) for temp, ph mid point and orp - see https://github.com/virgilm/raspipool/blob/master/calibrate.py, but it only works with I2C sensors for now. Thanks for the project @segalion, I still have a lot of things that don't fully work.

obert0, 2022-07-12:
I make some search and I was wondering if the official calibration method for ezo board are quite the same as the sensor filter from Homeassistant?