segalion/raspipool #16: Injector

helix0712, 2020-10-01:
Hi,

appreciate your project! Started almost the same on ESP8266, running MQTT and NodeRed on a central Synology Docker Instance. Currently in the process of rewiring the prototype and mounting everything on DIN rails. 
Your bypass looks nice and i will most probably use it in my installation. Now that i have pH, ORP and pump parts i was wondering how i would be able to inject the liquids (like PH -). Have you found a part to connect the silicone hoses safely to the 50mm pipes?

Thanks for sharing your project!

Patrick

andyyyy2306, 2020-10-01:
Hi Patrick,

could you share your Project on ESP8266 ?
Andyyyy

helix0712, 2020-10-01:
I am working on a git repository...but it may take some more time

andyyyy2306, 2020-10-05:
perfect ! thank you !

segalion, 2020-10-07:
Sorry about so low level of support during last year...
To inject muriatic acid and bleach, you need a valve injection inside a collar attached to the returning pipe:
I will put the first spanish reference that I can found in Internet (please only for reference...)
Valve injection:
https://www.poolaria.com/collarines-de-toma-y-pinzas/647-collarin-de-toma.html#/2756-medida-o_50_x_
collar pipe:
https://www.poolaria.com/recambios-piscinas/2662-valvula-inyeccion-completa-dosificacion-astralpool-4408030121.html

Remember distance between them >=50 cm from sand filter, and between both injections if you want or plan simultaneous injection