segalion/raspipool #18: Question: Is ORP measurement reliable outdoors with CYA present?

matt-sullivan, 2020-11-11:
I'm considering building a raspipool system, but I've read that ORP measurement is [difficult](https://www.aquaticsintl.com/facilities/maintenance/understanding-orp-and-ppm_s) [in outdoor pools](https://www.troublefreepool.com/threads/why-are-cya-levels-lower-when-using-orp.13287/) with cyanuric acid buffering the chlorine levels.

Can anyone confirm how difficult / problematic ORP measurement is? (I don't need commercial quality results, just looking to save time at home.) If the measurement is very dependent on sunlight, has anyone tried only measuring ORP at night time?

segalion, 2020-11-12:
There is much controversy with ORP as a measure of free chlorine. In "European school" it is well accepted, even with CYA. The "American school" the ORP is not very well regarded.  I am not an expert, and I would refer you to professional forums for these types of questions.

From what I have been able to read and understand, the ORP is closer to what you really want to measure as "disinfection quality", rather than the amount of disinfectant.

It could be said that the ORP measures the speed with which the water is being disinfected, supposedly because there is free chlorine and organisms to be eliminated. But the measurements can be altered by other factors (type of water, etc.) so it is very convenient to make a real adjustment between free chlorine and ORP values ​​in each pool, from time to time.

To give an example, we could have a low ORP and a lot of chlorine, perhaps because the water is at 5ºC, in a dark environment (covered), which prevents organisms from reproducing. Thus there is a lot of "oxidizer-reducing" potential, but nothing to "oxidizing-reducing".

CYA dramatically reduces the effectiveness of chlorine, which is why it is so important to maintain CYA ratios.

The only problem I see with ORP measurements is the behavior of the curves, where little change in ORP means a lot change in chlorine. With the CYA the only thing that happens is that these curves will be even less pronounced

Lauriekr, 2020-11-24:
I agree with Segalion, ORP is a betterindicator of disinfection potential as long as the value is above 600mV than a chlorine residual if the pH is constant. The CYA stabilizes free chlorine but in doing so it is a less powerful oxidant, although CYA is an acid, it's very weak and has little effect on pH, temperature has a bigger effect on ORP readings butif temp is meansured it can be corrected in the controller.

matt-sullivan, 2020-11-27:
Thanks @segalion & @Lauriekr.

I understand the theory of ORP vs free chlorine but I was more concerned how practical ORP is for DIY automation (with CYA present.) I.e. Are the sensors accurate enough and the curves far enough apart to meaningfully monitor/control chlorine without reacting to noise more than real changes in water chemistry. (One of the links I included also mentioned CYA & ORP levels change due to sunlight, so I wasn't sure if that introduced more noise/errors into the system.)

You haven't raised any of those concerns so I'll go ahead and buy some sensors and start tinkering.

closed 2020-11-27 by matt-sullivan

Lauriekr, 2020-11-27:
I didnt raise any concerns because I dont know for sure, you're right to hold concerns because they could be factors. AllI would say is that ORPis a much more robust measurement technique that residual chlorine of pH. Good luck, I will be doing same in a fw months time so please keep us updated.