segalion/raspipool #19: Question about the bypass for sensors

uclladt, 2020-11-12:
Why have you placed the bypass for the sensors between de pump-input and the filter output ?
Is there a specific reason for that ?

In my case i would like to place the bypass sensors after the filter output. 
Could it be a problem ?

segalion, 2020-11-12:
It is not the best place, because:
1- high flow of water can make meassurements not as good as in bypass with less flow. But some people reported good quality meassurement.
2- Maintenance. If you have to change probes, or you have a leak water... you have to close all circuit, and cant filter until fixed. With bypass, you can close it and work with main circuit as usual.
3- You have to ensure the probes are allways inside wateer. If not running, maybe you get the probes dry, and probably go bad in a few time. Thats why bypass is better as bottom as possible.

But, except for this... if you are careful with this, I think that you can do it.

equality1221, 2023-05-25:
From my understanding of the original question, uclladt was not suggesting no bypass at all, and was not suggesting placing the probes directly into the filter output line, and was not suggesting a physical location higher than your suggestion of "as low as possible".

Instead, the question was, why bypass the pump and filter?... as opposed to simply adding the exact same type of bypass piping into the filter output line.

This allows for all of the advantages of your design, e.g.:

a bypass with less flow (and pressure), thanks to the multiple changes in pipe diameter;

a bypass for maintenance purposes, so that you can shut off the valves to the piping that houses your probes - in the event of needing to change out probes, or to work on the filter or pump - you are able to isolate the two systems;

a bypass that is set a low as possible, lower than the filter output line, so that the probes always stay submerged in water, even if the pump is not running.

However (and this is the crucial point here), by putting this bypass in the filter output line on the way back to the pool, you don't have water needlessly flowing past the filter, going untreated.

This design is superior to the one that you originally suggested, because yours is designed to purposely let water bypass the filter, and therefore go untreated. Even if it's not much water, that is a needlessly unnecessary design choice.

The question now becomes: "would you rather be checking the status of the water that is being returned to the pool, or the status of the water that is coming from the pool?"

The answer to this question might be a personal choice. I think it's probably best to be checking the water that is coming from the pool.

Your design is testing the water coming from the pool, which likely gives a more accurate temperature reading (as temperature is likely to change within the pump, filter, and associated piping, giving a somewhat different value than that of the actual pool itself), and possibly even a more accurate reading of chemicals (since the filter could sightly change the measured values of the chemicals to differ from the water coming directly from the pool).

In order to get the best of both worlds, this same bypass setup could be placed into the line coming from the pool, somewhere before the pump, and going back into that same line.

"Bypass" in this sense is to say that you are bypassing part of the pump supply piping, for the purposes as stated above: decreased flow and pressure for better readings; maintenance shutoffs; and continuous submersion. It is not bypassing critical parts of the pool's system, it's not bypassing filtration, it's not really bypassing anything. It is a bypass strictly for its own sake.

jonpcar, 2023-07-13:
Hey equality, you are misunderstanding the water flow through segalion’s bypass.  It is NOT bypassing the filter.  At the filter output, water pressure is higher than at the the pump input.  Therefore flow of water through the manifold/bypass will be from water that is filtered (via the filter output) back into the pump…and thus run through the filter AGAIN. 

On segalion’s diagram, flow through the “bypass” (sensor manifold), is from right to left:
https://github.com/segalion/raspipool/wiki/Bypass-for-sensors