segalion/raspipool #23: Atlas Scientific Tentacle T3

SiPhyLLC, 2021-03-15:
Has anyone tried using this RPi HAT (Tentacle T3) [https://atlas-scientific.com/electrical-isolation/whitebox-labs-tentacle-t3/#](url) from Atlas Scientific?  I haven't yet explored if the connections are conflicting with the Relay Board or if the code would have issues.  It would make for a cleaner installation, and add an additional EZO location.

huehueteotl, 2021-05-28:
Hi @SiPhyLLC, I use a T3 for my setup: see closed topic https://github.com/segalion/raspipool/issues/10

danielegger1, 2022-01-30:
Hi
I read through the other post, great work :)
did you had timeluck to implement the pumps with I2C?

huehueteotl, 2022-01-31:
Hi @danielegger1 
Always nice to hear there are other people with the same interests 😄 

As a matter of fact I abandoned this project mostly because of the (really) bad performance and stability of HA on the RPi!
Because I have a dedicated old Mac Mini running Ubuntu with HA to control the rest in the house, I opted for a lighter client in the pool and bought the Atlas Pool Kit (https://atlas-scientific.com/kits/bb-wi-fi-pool-kit/).

This simply connects with MTTQ (also running on the "macbuntu") to update the sensor readings and I did some hardware mods to control a relais for my pool pump, light and some extra sensors to read internal/external temperatures.
It's much more stable, quicker to restart and a lot easier to maintain and configure in HA 😉 

But to answer your real question: nope, I still have not connected the pumps...

danielegger1, 2022-01-31:
Hi thanks for the quick reply :)
Good point, I looked originally into the Wifi kit, but thought doing it through the Pi is better, but purchasing the kit will allow me both ways, instead of the Tentacle I can wire the 4 Cables and can try both ways :)
Going through MQTT is something I have never worked with, but should be possible todo, I found some examples for Thingspeak (I think you mean going through that, right?)
Do you have some images of your setup / HA screen? I am interested in your design :)
There is also this interesting project available:
http://rws.casa-eller.de/modus.html
Which could help a lot, but has the same original problem, no I2C Pump possible.
I just figured out (during writing with you), that the wifi pool kit can even directly connect the pump, so thats a cheap out of the box working unit, great :)

so thank you, you really helped me a lot!

CrociStrike030, 2022-02-09:
> Hi thanks for the quick reply :) Good point, I looked originally into the Wifi kit, but thought doing it through the Pi is better, but purchasing the kit will allow me both ways, instead of the [...]

Hi! 
So if i understand correctly, the WiFi poolkit does everything by itself and can also control the filter pump and chlorine pump?

huehueteotl, 2022-02-11:
Hi @CrociStrike030,
Technically and theoretically yes: the Atlas WiFi poolkit can control everything.

But the basic code provided by Atlas (https://github.com/Atlas-Scientific/Ezo_I2c_lib/tree/master/Examples/IOT_kits/pool_kit) reads the sensors you've attached (pH, ORP and RTD) and published the values to thingspeak.com using MQTT.
It can control one pump based on the pH value and static threshold and dose values you set.

In my setup I replaced thingspeak with my own local Mosquitto server, and beside the PUB channel I added a SUB channel so I can send commands back to the Huzzah instead of just using serial commands (clearly just a demo from Atlas).
Also added the ESPAsync WebServer for OTA updates (must-have for headless ESP boards!).

To be able to control my pump and light I slightly modified the board to have enough I/O ports for what I want.
I removed the enable/disable pins (just clipped them off at the Huzzah socket connector) and rewired them to my external relais board, and a DHT111 sensor. Also had a MCP9808 sensor lying around which is i2c so easy to add.

TL;DR version: you will need to do some hardware mods and write your own sketch to control everything.