segalion/raspipool #26: Probe placement question

sddgit, 2021-09-14:
I’m not sure if anyone’s monitoring this project anymore, but here goes. Why are the sensors recommended to be installed in a “bypass” pipe? Can’t they just be installed in the existing pool plumbing?

Goose1451, 2021-09-14:
Still here... I haven't installed my probes yet, but have planned the install.  And I had the same question as you.  I'm planning the install on my spa.  I'm planning to put the probes on the plumbing coming from the filters to the circulation pump so I am getting readings of water coming from the tub, rather than heated and going to the tub.

I honestly think the bypass is for ease of maintenance incase you wanted to isolate that branch of plumbing.

SiPhyLLC, 2021-09-14:
I believe someone asked this before and the answer was that the water in
the main lines is turbulent and results in inaccurate measurement values.
The bypass slows down the flow and provides for better measurement accuracy.

sddgit, 2021-09-14:
Oops, don’t know how I missed the other issue! I am tempted to try just on the “suction” pipe from my spa. Hopefully the pressure won’t be too overwhelming…

oylern, 2021-11-22:
Curious to hear if you had any issues adding the probes to your suction line or if it worked just fine. Any updates?

sddgit, 2021-11-22:
I'm still in the process! Got one working for a while in the normal pool plumbing, just after the filter. It worked for a little while, but then the Sonoff TH16 started to turn itself off. Took the probe out, and the wiring sleeve had been damaged during installation. I’ve ordered some new ones and will try again soon. I'll update then.

sddgit, 2021-11-30:
OK - it’s in the main suction line, and has been running fine for a couple of days. Temperature seems accurate and stable.