segalion/raspipool #28: I know To little

bbombek, 2022-01-30:
Hello!
Last year I went into this project. I bought all the BUM. Yesterday I did some progress and now I am stuck.
So I know to little on the programming side of RPI HA.
What I did...
SOlder the EZO circuits on carrier board, connected the thing per your scheme.
Installed HA operating system 64bit via this tutorial  https://www.home-assistant.io/installation/raspberrypi
Copied all the project files as the instructions were.
**1. Now the first problem is that when I check configuration it says**

[image]

**2. Then the lovelace looks like this**

[image]

**3. I dont know coding but somehow it doesn't read the sensor data from the atlas_scientific custom components folder...
Should I do something with this since it look like this? (example.yaml filename is ok??)**

[image]

**4. This is how the connection looks like:**

[image]

I sure am missing something basic but I just dont know.  Can someone please help me, since I bought all this materials and now I dont know how to make them work.
Thank you
Boris

virgilm, 2022-01-30:
Did you install HA as a container or via python venv? Either way, the custom component seems to be in the right place.
Check if RPi can read from your sensor (https://files.atlas-scientific.com/pi_sample_code.pdf). Then if all that is correct, check your HA logs and post what you are seeing.

Adding this in your configuration.yaml might help you debug further:

```
logger:
  default: info
  logs:
    custom_components.atlas_scientific.sensor: debug
```

bbombek, 2022-02-01:
Hello virgilm!
Thank you for the quick response.
Installed HA operating system 64bit via this tutorial https://www.home-assistant.io/installation/raspberrypi
So no raspberry pi OS
The atlas scientific pdf you sent i cannot get to the config (sudo raspi-config command) since it is a HA OS.
my configuration.yaml is like this so the debug for atlas scientific sensor is there.
Any suggestions or did I do everything wrong...

# Configure a default setup of Home Assistant (frontend, api, etc)
# default_config:

# Text to speech
# tts:
#  - platform: google_translate

# group: !include groups.yaml
# automation: !include automations.yaml
# script: !include scripts.yaml
# scene: !include scenes.yaml

homeassistant:
  name: RaspiPool
  # Location required to calculate the time the sun rises and sets
  latitude: 46.4453269259808
  longitude: 15.857537091648812
  unit_system: metric
  customize: !include customize.yaml
  packages: !include_dir_named packages
default_config:
# Disable Discover some devices automatically
# discovery:
tts:
  - platform: google_translate
logger:
  default: critical
  logs:
    homeassistant.components.sensor.atlas_scientific: debug
    homeassistant.components.switch.command_line: debug

group: !include groups.yaml
# Disable automation
# automation: !include automations.yaml
script: !include scripts.yaml
lovelace:
  mode: yaml

virgilm, 2022-02-01:
Please format your file...
Well, that's because HAAS or HA OS or whatever is called nowadays is a joke! (at least on RPi). You can try [this](https://github.com/adamoutler/HassOSConfigurator) to make sure serial is configured. Send the logs (from your config/.home-assistant.log). Good luck!

bbombek, 2022-02-01:
Sorry for the format I copied and pasted it and that is what turned out :(
I edited the upper post.
So you suggest to install raspberry pi OS and then the HA container?

[image]

here is the log.
[home-assistant.log](https://github.com/segalion/raspipool/files/7980875/home-assistant.log)

bbombek, 2022-02-01:
Hello again
This I dont find in the addon store on my HA OS. Maybe because it is not HAAS anymore?
Confused...
https://github.com/adamoutler/HassOSConfigurator

virgilm, 2022-02-01:
Can't clear that confusion for you, HA made it as hard as possible to figure it out... I'm not suggesting anything about HA, except that is evolving in the wrong direction (at least for this project).

You can fix your problem at hand by adding the following line to your custom_components/atlas_scientific/manifest.json
`  "version": "1.0.0",`. Another breaking change in HA - they don't care about backward compatibility one bit.

That will get you unstuck at least. I have a fork here: https://github.com/virgilm/raspipool, and I will try to bring it up to date, but the interest in this project seems to have waned...

bbombek, 2022-02-01:
Hmm
tried the line nothing happened
I overwrote your fork put in the longitude and lattitude, pushbullet and weather APIs and now this happens
[image]

[image]

Stuck here.
I want to write a tutorial for all of this step by step on this date and I hope its not that difficult... anyway... :)

virgilm, 2022-02-01:
Don't get my fork... it has some components that need more support, and I'm not sure I'm ready to support that fork :)

Just change your manifest file to read:
```
{
  "version": "1.0.0",
  "domain": "atlas_scientific",
  "name": "Atlas Scientific",
  "documentation": "https://github.com/home-assistant/example-custom-config/tree/master/custom_components/example_sensor/",
  "dependencies": [],
  "codeowners": [],
  "requirements": ["pyserial", "smbus-cffi==0.5.1"]
}
```

bbombek, 2022-02-02:
HMM i did that and still problems... the same

bbombek, 2022-02-05:
Does maybe anybody have the SD-card Image for raspberry that just works? Because I cannot get that to work.
PLEASE...!

virgilm, 2022-02-05:
I put up my latest working fork at https://github.com/virgilm/raspipool. Your sensors would have to be in I2C mode and the installation is not trivial. Good luck!

bbombek, 2022-02-06:
OK I will try this hope it works! How do I set sensors in I2C mode?
Thank you