segalion/raspipool #30: Bypass probe

YannGranger, 2022-07-09:
Hi, 
Very interesting project. I was wondering why do you have to have a bypass for the probe. 
can’t they be installed in line on the existing pipes ?
Thanks

YannGranger, 2022-07-09:
Sorry this topic has already been covered in previous issue. 
i close the issue

closed 2022-07-09 by YannGranger

SiPhyLLC, 2022-07-09:
> There is generally too much turbulence in the main line to get accurate
> measurements.  Having a branch creates a less turbulent path and more
> steady probe measurements.

Oops!  Sent to soon.

The branch also makes it easy to replace the probes as you can isolate the
branch from the rest of the system by closing off valves on both ends of
the branch.

SiPhyLLC, 2022-10-11:
There is generally too much turbulence in the main line to get accurate
measurements.  Having a branch creates a less turbulent path and more
steady probe measurements.