segalion/raspipool #32: Specifics on Plumbing fittings

Brikpilot, 2022-09-26:
Hi
I am uncertain on some of the plumbing parts to be used. 

One is solved for Tee pieces, they are called Hough Section Connectors.  (some countries may not use these)
https://www.aliexpress.com/item/4000540605421.html?spm=a2g0o.cart.0.0.2bc938da0YTkZV&mp=1

Can anyone help with what to mount the sensors in please as the plumbing  downstream of the pump comes under rather high pressures.  This will require a good seal between pipe and sensor.   Another consideration is that  these sensors will eventually fail so should be relatively easily replaced.  Appreciate being pointed to any fittings suitable that will achieve this..
Thanks ahead

STELTZ-MANIA, 2022-09-26:
Hi,

The parts I used to mount the probes are:

PVC Pressure Faucet Tee
https://www.holmanindustries.com.au/product/pvc-pressure-faucet-tee/

and a Cable gland like this:
https://www.voltexelectrical.com.au/p-3057-nylon-cable-gland-20mm-cable-od-7-14mm-25-pack.aspx

I only finished the plumbing 3 days ago but so far there are no noticeable leaks and should be easy to replace in the future.

Regards

Brikpilot, 2022-09-26:
Thanks very much for the details.  The cable gland was what I could not figure (from online data) to match pipe thread to sensor diameter and be snug and leak free under pressure. Also great to see an example (and presumably in Oz based on Deta & Holman brands) :-)

Wiring on LHS of filter, is that a temperature sensor?  I note you have  a third tee piece for a sensor, You could use this  for temperature too.  
I made my own using a Sonoff DS18B20 Sensor and TH10 to read it
To mount it I started with https://www.holmanindustries.com.au/product/pvc-pressure-screwed-plugs/
Drilled a hole in the centre, ran temperature probe wiring through with a lot of heat shrink protection, then infilled with hot glue.  No leaks and didn't fry the probe making it.  Did have to drop it in freezer to help fast set the hot glue.  Note I hot glued over the wire not the probe.  On your setup you could readily screw this in, however the tip of the probe will protrude about a third into the middle of the flow.

Thanks again

STELTZ-MANIA, 2022-09-26:
The wiring on the LHS of the filter is for the WiFi adaptor for the Heater. This has never worked for me and is currently disconnected. One day I will disconnect the cable from the heater as well. Yes indeed, the third T-piece will be for a DS18B20 temp sensor, there is also a 4th T-Piece on the return line to the pump which is out of site where there will be another DS18B20 so that I can see the temp on the way in and out.

I ended up buying every size cable gland from Bunnings to work out which one fit best. Unfortunately they don't list them on their website at the moment so can't link to the exact one I used.

The other thing just out of site is a 3-way jandy valve at the bottom of the probe pipe to enable bypass and also regulate the flow because i didn't fully trust the gland under full pressure but it seem to be holding up fine.

I also never intended for the piping to go over the top of the powerpoint but I ran out of space and it gives me the heebies...