segalion/raspipool #38: I cannot create the boot/config.txt file.

Extesat, 2024-10-20:
Does the project actually work?
I cannot create the boot/config.txt file.
Can you help me to create the file?
I am using Raspberry 5 with the following operating system, installed with IMAGER
Other specific-purpose OS -> Home assistant and home automation -> Home Assistant -> Home Assistant OS 13.2 (RPI 5).
Thanks
------- Translated from Spanish to English with Deepl.com-----
¿El proyecto funciona actualmente?
Yo no puedo crear el archivo boot/config.txt
¿Puedes ayudarme a crear el archivo?
Yo utilizo  Raspberry 5 con el siguiente sistema operativo, instalado con  IMAGER
Other specific-purpose OS -> Home assistant and home automation -> Home Assistant -> Home Assistant OS 13.2 (RPI 5).
Gracias

cremer42, 2024-10-20:
I got it working somewhat early last year. It was a lot of work and there wasn't any help from this page. I had to do a lot of trial and error. 

My recommendation is to steer clear of the rpi5 and get a less expensive option like an ESP32 relay board. I just completed my own, I've not had time to clean up the code and create a repo for it yet.  But it's in my plans.

cremer42, 2024-10-21:
> Does the project actually work? I cannot create the boot/config.txt file. Can you help me to create the file? I am using Raspberry 5 with the following operating system, installed with IMAGER Other [...]

If you are deadset on using the rpi.... I can send you what i have. I dont know the differences between the rpi4 and rpi5, but up to a month or so ago, I had it this working. I was using atlas scientific sensors via UART. A few temperature sensors. My setup was for a 120v pump with single speed, I had a secondary pump for a vacuum and I also controlled a pool heater, but I could help you get this stuff corrected for your setup. Just lmk and I could send you all my files/code.

cremer42, 2024-10-21:
> Does the project actually work? I cannot create the boot/config.txt file. Can you help me to create the file? I am using Raspberry 5 with the following operating system, installed with IMAGER Other [...]

I actually forked this repository and updated all of the code with my working code. I included my config.txt file that was working in my rpi4. I am unsure this will work in rpi5, but hopefully for you it will. Let me know if you have any further questions.

De hecho, bifurqué este repositorio y actualicé todo el código con mi código de trabajo. Incluí mi archivo config.txt que estaba funcionando en mi rpi4. No estoy seguro de que esto funcione en rpi5, pero espero que para usted sí. Déjame saber si tienes más preguntas.

Extesat, 2024-10-21:
Hi cremer42
I am indeed at an impasse with this project, I can't create the /boot/config.txt file.
Homeassistant does not recognise the atlas scientific sensors nor the temperature sensors.
I thank you for your offer to send me your code, I can test if it works on the rpi 5.
Thank you very much
Regards

cremer42, 2024-10-21:
> Hi cremer42
>
> I am indeed at an impasse with this project, I can't create the /boot/config.txt file.
>
> Homeassistant does not recognise the atlas scientific sensors nor the temperature sensors.
>
> I [...]

For sure, check my page, I have a forked copy of the raspipool system with my working code, including the config. If you have any questions let me know and I can help you out along the way. Just start new "issues" on my fork.

STELTZ-MANIA, 2024-10-21:
Its not straight forward but it does work. I've been running for about 2 years now using EZO-pH and EZO-ORP and it seems to be functionality without any real issues. I'm about to try and add in an EZO-FLOW.

I am using a Raspberry Pi 4, a Whitebox T3 and 3 1-Wire temperature probes.

[image]

cremer42, 2024-10-21:
> Its not straight forward but it does work. I've been running for about 2 years now using EZO-pH and EZO-ORP and it seems to be functionality without any real issues. I'm about to try and add in an EZO-FLOW.
> 
> 
> 
> I am using a Raspberry Pi 4, a Whitebox T3 and 3 1-Wire temperature probes.
> 
> 
> 
> [image]
> 
> 

Is there a reason you're concerned on flow and not pressure? I've got pressure on my new setup, I'd rather see the pressure to tell me when I need to backwash my filter. [image]

STELTZ-MANIA, 2024-10-21:
The main reason I am interested in Flow is to ensure that the Heater is working at optimum efficiency. The specification for the heater is 125L/min but at the moment I really have no idea what the flow rate is. The filter has a pressure gauge so I can eyeball that every couple of days, therefore It wasn't as high a priority. 

Also, my pump is located higher than the pool with a non-return valve but every now and then it doesn't self prime (i think sometimes a bit of debri gets caught in the valve). During week days the pump will turn on at 11am when I am in the office so I wated a way to ensure it was actually pumping.

cremer42, 2024-10-22:
> The main reason I am interested in Flow is to ensure that the Heater is working at optimum efficiency. The specification for the heater is 125L/min but at the moment I really have no idea what the [...]

That makes sense... I guess i never worried about that. My pump sits below my skimmer intake and my pump moves about 200 L/min. (unless my filter is clogged) ie my reason for pressure, I know my average pressure with a clean filter, filter dirty, then i know my flow is going to be reduced. lol

Extesat, 2024-10-23:
> For sure, check my page, I have a forked copy of the raspipool system with my working code, including the config. If you have any questions let me know and I can help you out along the way. Just [...]

I have installed your Fork and although it works, it still does not detect the sensors. I think the problem is that I am unable to find where to install the /boot/config.txt file.
The SD card has 8 partitions, but only mounts 4 partitions.
I have mounted partition 1 from Linux. The system automatically names it Hassos-boot.
I copied the config.txt file and now it contains:
slot-A (directory)
cmdline-tryboot.txt
cmdline.txt
config.txt
tryboot.txt
Nevertheless, your fork does not detect the Ph, ORP and temperature sensors.
Can you tell me where should I put the config.txt file for raspipool to detect the sensors?
Thanks

cremer42, 2024-10-23:
[image]
 Maybe its different for a pi5, but my SD card is named "hassos-boot" and has the file structure in the attached picture.